maniacx/Battery-Health-Charging #133: Asus ProArt PX13/Nobara/Kernel 6.13.2

alzv-adkey, 2025-02-10:
It was working fine on system with 6.12 kernels? but after update it shows unsupported device message.

alzv-adkey, 2025-02-17:
1. ls /sys/class/power_supply/
AC0  BAT0  ucsi-source-psy-USBC000:001  ucsi-source-psy-USBC000:002

2. ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
ls: cannot access '/sys/class/power_supply/BAT0/charge_control_end_threshold': No such file or directory

maniacx, 2025-02-17:
Seems like the kernels asus wmi module is probably is not working for you. As you can see, it doesn't exposes charge threshold sysfs nodes.

> ls -l /sys/class/power_supply/BAT0/charge_control_end_threshold
> ls: cannot access '/sys/class/power_supply/BAT0/charge_control_end_threshold': No such file or directory

Are you use Nobara ship with 6.13 or did you manually upgrade the kernel.

alzv-adkey, 2025-02-17:
Nobara ships with 6.13.2 kernel. I was using only nobara upgrade tools with no custom repos

alzv-adkey, 2025-02-17:
lsmod | grep asus_wmi
asus_wmi              143360  2 asus_armoury,hid_asus
platform_profile       12288  2 amd_pmf,asus_wmi
i8042                  57344  1 asus_wmi
rfkill                 40960  8 asus_wmi,bluetooth,cfg80211
sparse_keymap          12288  1 asus_wmi
video                  81920  3 asus_wmi,amdgpu,nvidia_modeset
wmi                    32768  4 video,asus_armoury,asus_wmi,wmi_bmof

alzv-adkey, 2025-02-17:
In 6.12 the difference only in that line

asus_wmi              143360  3 asus_armoury,asus_nb_wmi,hid_asus

6.13 has no "asus_nb_wmi" loaded

alzv-adkey, 2025-02-17:
I've modprobed "asus_nb_wmi" module manually and everything works now. Many thanks!

maniacx, 2025-02-17:
Great. Do report to nobara about the kernel config.

closed 2025-02-20 by maniacx