maniacx/Battery-Health-Charging #134: UX not shown anymore on Gnome 46

rene-hermenau, 2025-02-27:
The settings can still be opened on the gnome extension website and it shows that the battery threshold is active but it's not working anymore and battery is constantly charged. Also, in gnome 46 the UX elements for opening the settings modal are gone.

maniacx, 2025-02-27:
Also try disabling and enabling the extension. Most likely if there is an issue you should see notification about it.

maniacx, 2025-09-22:
@Zathrus 
What is your polkit version?

```
pkaction --version
```

Let's verify if rules are installed correctly. 

Open terminal / console
Note: we will use sudo -s command to read the rules. So do not use any other commands. Any command use to write / or delete may make the system unusable

```
sudo -s #root mode

#check permission of directory
ls -ld /etc/polkit-1/rules.d

# cd
cd /etc/polkit-1/rules.d

# list all files in contents rules.d
ls -l /etc/polkit-1/rules.d

```

This should contain a file name something like with your user name. Example if usernane is abczyz
10.dem.batteryhealthcharging.setthreshold-abcxyz.rules

If file exist read it using cat command file name
```
cat 10.dem.batteryhealthcharging.setthreshold-abcxyz.rules
```

content should be something like
```
polkit.addRule(function(action, subject) {
    if (action.id === "org.freedesktop.policykit.exec" &&
        action.lookup("program") === "/usr/local/bin/batteryhealthchargingctl-abcxyz"&&
        subject.user === "abcxyz")
    {
        return polkit.Result.YES;
    }
});
```

Also check if tool is installed 
```
ls -l /usr/local/bin/batteryhealthchargingctl-abcxyz
```

If these both files exist and your contents of rules files are correct. It should not ask for password.

Do these checks and let me know.