maniacx/Battery-Health-Charging #135: Update broke extension functionality? (Dell platform).

natus-w, 2025-03-05:
**Laptop Type:**Dell Inspiron 7370

**Kernel build:**6.11.0-14-generic

**Linux build:** Ubuntu 24.10 (GNOME/Wayland)

Issue

Extension was working previously (used libsmbios backend), however after the update to 74 it seems to no longer be working.

The first symptom was an extended pkexec timeout which would cause the extension to peg CPU usage.

I then followed the mitigation steps as described in #122 (have sudoers policy in place, manually edited the corresponding lines), however now the extension is asking for a polkit update which doesn't want to apply.

[image]

maniacx, 2025-03-06:
Oh, you may have to patch the extension again.

Or just open the extension folder

And replace the text  `pkexec` with `sudo` in file /lib/driver.js and /lib/helper.js

Next release , I will bake this in the extension instead of manually patching.

Apologize for the inconvenience.

natus-w, 2025-03-07:
> Oh, you may have to patch the extension again.
> 
> Or just open the extension folder
> 
> And replace the text `pkexec` with `sudo` in file /lib/driver.js and /lib/helper.js
> 
> Next release , I will bake this in the extension instead of manually patching in the next release.
> 
> Apologize for the inconvenience.

Thank you for your reply.

The error still persists after making the required edits (manually in the corresponding files)

[image]

[image]

I have tried reinstalling from script but the error persists (somehow - I am not sure if I've skipped a few steps or not..)

maniacx, 2025-03-07:
Try running
```
sudo ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/tool/installer.sh  --tool-user $USER install
```

natus-w, 2025-03-07:
> Try running
> 
> ```
> sudo ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/tool/installer.sh  --tool-user $USER install
> ```

Just tried that now, unfortunately the Polkit error still persists when the extension is re-enabled (via Extension Manager).

maniacx, 2025-03-07:
Thats strange.

When trying command can you post the result as well of the command.
Also when using these commands make sure the terminal is not opened in superuser/root mode. As in do not so use `su` or `sudo`
Run these commands in terminal as user.

Check if root or user
```
whoami
```
it should output should be your user name. If it showing root, mean you are in superuser/root mode

Remove polkit rules. Post your output of these command.
```
sudo ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/tool/installer.sh  --tool-user $USER remove
```
Install polkit rules again.  Post your output of these command.
```
sudo ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/tool/installer.sh  --tool-user $USER install
```

Check the result of CHECKINSTALLTION command. WHen running this command, note if sudo is asking for password. If sudo rules have been applied correctly, sudo shouldnt ask for password here. Let me know if ask for password or not. Post the result of this command too

```
sudo /usr/local/bin/batteryhealthchargingctl-$USER CHECKINSTALLATION /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources $USER
```

natus-w, 2025-03-09:
[image]

Removing polkit checks out fine, as does install.

[image]

However the problem still persists after the extension is reloaded. I am going to try restarting the machine in between wiping the polkit options and see if that fixes anything..

[image]

maniacx, 2025-03-09:
I have modified the command to get some more info.

Open a new terminal window and paste this command and post the output here

```
echo "Current user: $(whoami)"; timeout 0.3 sudo /usr/local/bin/batteryhealthchargingctl-$USER CHECKINSTALLATION /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources $USER; echo "Exit status: $?";
```

maniacx, 2025-03-09:
Open another  terminal window.
```
sudo visudo -c -f /etc/sudoers.d/batteryhealthcharging-$USER
```

post output here

natus-w, 2025-03-11:
> Open another terminal window.
> 
> ```
> sudo visudo -c -f /etc/sudoers.d/batteryhealthcharging-$USER
> ```
> 
> post output here

This ended up being the problem - the visudo file managed to get wiped (somehow)

Replacing this file solved the polkit permissions issue, the program works as expected now.

[image]