maniacx/Battery-Health-Charging #141: Battery Health Charging Error - Encountered an unexpected error. (Dell)

zz0rk, 2025-03-13:
[image]

**Debian Bookworm
GNOME Shell 43.9
Battery Health Charging 73**

**Installed libsmbios from https://github.com/dell/libsmbios. This will install smbios binaries to**
`/usr/local/sbin/`

**Quote from your documentation:**
_The extension checks if smbios-battery-ctl from libsmbios is installed at
/usr/sbin/smbios-battery-ctl._

**Moved binaries to /usr/sbin**
`mv /usr/local/sbin/smbios-* /usr/sbin/`

**ls -l /usr/sbin/smbios-battery-ctl**
`-rwxr-xr-x 1 root root 11490 Mar 12 16:12 /usr/sbin/smbios-battery-ctl`

**ls -l /sys/devices/platform/dell-laptop**
```
total 0
lrwxrwxrwx 1 root root    0 Mar 11 22:00 driver -> ../../../bus/platform/drivers/dell-laptop
-rw-r--r-- 1 root root 4096 Mar 12 16:28 driver_override
-r--r--r-- 1 root root 4096 Mar 12 16:28 modalias
drwxr-xr-x 2 root root    0 Mar 12 16:28 power
drwxr-xr-x 3 root root    0 Mar 11 22:00 rfkill
lrwxrwxrwx 1 root root    0 Mar 11 22:00 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Mar 11 22:00 uevent
```

**smbios-battery-ctl --get-charging-cfg**
```
Traceback (most recent call last):
  File "/usr/sbin/smbios-battery-ctl", line 275, in <module>
	sys.exit( main() )
			  ^^^^^^
  File "/usr/sbin/smbios-battery-ctl", line 241, in main
	print_primary_battery_cfg()
  File "/usr/sbin/smbios-battery-ctl", line 190, in print_primary_battery_cfg
	mode = get_charging_mode()
		   ^^^^^^^^^^^^^^^^^^^
  File "/usr/sbin/smbios-battery-ctl", line 140, in get_charging_mode
	active = table[mode.value].isActive()
			 ~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/libsmbios_c/trace_decorator.py", line 103, in trace
	result = func(*args, **kw)
			 ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/libsmbios_c/smbios_token.py", line 145, in getitem
	raise IndexError(_("SMBIOS Token ID 0x%04x not found") % id )
IndexError: SMBIOS Token ID 0x0341 not found
```

I am not sure what's wrong here. libsmbios seems to be install correctly.

Thank you

maniacx, 2025-03-13:
It depends on your laptop model. 

Libsmbios is a older pacakage opensource. It usually works well with older laptops. But doesn't with newer modrl, Specially xps model

Dell command center are compatible with new Dell laptop in it is dells proprietary package.

There is no documentation for package vs supported model. You just have to try both and use the one that works.

I would suggest, you install Dell command center. Since libsmbios doesn't work for you.

Install the deb. And test it out using command line. If it works,  than use the extension.

Extension also detects what is installed
 If both are installed the extension will provide the option to choose Dell command center or libsmbios. So choose the one that works.

zz0rk, 2025-03-13:
I've command center installed on a Precision 5520 and it works great. 
The one I am having the problem with is a Latitude 6410, an older model. I initially installed command center, which I suspected wouldn't work, and it didn't. That's when I installed libsmbios. BTW, I had both command center and libsmbios installed at the same time and the extension did not detect what was installed.
Thank you for your help.

maniacx, 2025-03-13:
> I've command center installed on a Precision 5520 and it works great. The one I am having the problem with is a Latitude 6410, an older model. I initially installed command center, which I suspected [...]

Thank for reporting. I will have a look at it.

s4vi0rodrigues, 2025-03-19:
I have the same problem. I use a Dell Latitude 3540 Corei7 1355u with BIOS version 1.20.0 (10, March 2025).

maniacx, 2025-03-19:
> I have the same problem. I use a Dell Latitude 3540 Corei7 1355u with BIOS version 1.20.0 (10, March 2025).

Kindly give some details of your system as there are 3 ways to control threshold. 

What kernel version are you using? 
If kernel is older, and doesn't support changing threshold natively, which package are you using?
Have you tried changing threshold using command line?

Read the documentation and let me know if you having trouble, I can guide you.

https://macs18max.github.io//Battery-Health-Charging/device-compatibility/dell

McThump, 2025-03-30:
I'm getting the same error. Tried both libsmbios and Dell Command Configure. 

Dell XPS 13 9310
Ubuntu 24.04.2 Kernel Linux 6.8.0-56-generic

Using the libsmbios commands does not produce errors but has no effect on charging. Using the Dell commands produces an error, which implies OpenSSL is missing, but it is installed:

root@AIT136:/home/patrick# pkexec /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Custom:70-80
/opt/dell/dcc/cctk: error while loading shared libraries: libcrypto.so.1.1: cannot open shared object file: No such file or directory

root@AIT136:/home/patrick# openssl version
OpenSSL 3.0.13 30 Jan 2024 (Library: OpenSSL 3.0.13 30 Jan 2024)

maniacx, 2025-03-30:
Which version of dcc did you download?

https://www.dell.com/support/kbdoc/en-us/000178000/dell-command-configure

There is 5.2.0 and 5.1.0 both are compatible with 24.04. Try both.

[image]

But take note that there were similar issue years back regarding,  a fix for libcrypto.so.1.1,  that broke the whole ubuntu system 
Do not try any crazy solutions from net.

McThump, 2025-03-30:
Version 5.1.0-6. I'll try 5.2.0 later today.

McThump, 2025-03-31:
I couldn't find a download of 5.2.0 for Ubuntu. Do you have a link to it?

McThump, 2025-03-31:
Yes, that's what I did. Seems like the problem is that DCC is looking for a library (libcrypto.so.1.1) which is part of OpenSSL 1.1, and is not included in the standard Ubuntu repositories. OpenSSL 1.1 was replaced with OpenSSL 3.0 back on Ubuntu 22.04.

maniacx, 2025-03-31:
I could only find something this as solution

https://www.dell.com/community/en/conversations/linux-general/dell-command-configure-breaks-openssl-and-other-things/647f9fa2f4ccf8a8de480dd8?commentId=67e1ed16348a5b47bbe2e8dd

https://www.dell.com/community/en/conversations/linux-general/dell-command-configure-cctk-v410-on-ubuntu-2204-libssl11-not-found/647fa22cf4ccf8a8de79a38c?commentId=64a3fb9bf4ccf8a8dea104f8

Most likely it should work, but I would only do this if you can backup your disk drives/data and have a easy way to restore them just incase.

McThump, 2025-03-31:
Thanks, but installing an unsupported SSL library is too much of a security risk for me. I have reported the problem to Dell and will let you know what happens.

luyatshimbalanga, 2025-04-06:
I encountered a similar issue with Dell Inspiron 14 7425 2-in-1 after enabled the extension. 

Running  `smbios-battery-ctl --get-charging-cfg` led the following result

```
ERROR: Could not parse system SMBIOS table.

Common problems are:

    -- 'SMM Mitigations' is enabled in BIOS setup.
        Run kernel 4.15 or later with 
        dell-smbios-wmi enabled.
         or 
        Disable 'SMM mitigations' in BIOS setup.

    -- Insufficient permissions to perform operation.
       Try running as a more privileged account.
          Linux  : run as 'root' user
          Windows: run as 'administrator' user

    -- dell-smbios-wmi driver not loaded 
       Try loading the dell-smbios-wmi driver 
          Linux : modprobe dell-smbios-wmi 

    -- Call filtered by Linux kernel 
       Some functionality is natively supported 
       by the Linux kernel 

    -- dcdbas device driver not loaded.
       Try loading the dcdbas driver
          Linux  : modprobe dcdbas
```
Dell Command Center 5.1 is installed  which displays the battery charging mode

[image]

It seems bios password is required to switch to different modes which is not set on the system. Is there a better alternative
as I wish to no set up a bios password?

Here is the following system information:

## Hardware Information:
- **Hardware Model:**                              Dell Inc. Inspiron 14 7425 2-in-1
- **Memory:**                                      32.0 GiB
- **Processor:**                                   AMD Ryzen™ 7 5825U with Radeon™ Graphics × 16
- **Graphics:**                                    AMD Radeon™ Graphics
- **Disk Capacity:**                               2.0 TB

## Software Information:
- **Firmware Version:**                            1.21.0
- **OS Name:**                                     Fedora Linux 42 (Design Suite Prerelease)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               48
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.14.0-63.fc42.x86_64

maniacx, 2025-04-06:
If you havent set a bios password. It dcc or the extension shouldn't ask for a Bios password.

Can you try setting threshold using dcc in command line and post the output here.
1. Disable the extension
2. Set threshold using command line
3. Post exit code using echo $?
4. Read threshold using command line
5. Post exit code
Try differrent modes. 

Post the output of command and exit code here

```
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Custom:55-60
echo $?
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg
echo $?
```

Readme
https://maniacx.github.io/Battery-Health-Charging/device-compatibility/dell

luyatshimbalanga, 2025-04-06:
I confirm netiher dcc nor extension ask for password. Here is the requested output:

```
 sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Custom:55-60
echo $?

Error in Setting the Value.
62
```
```
sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg
echo $?

Usage error.

Dell Command Configure Version 5.1.0  23 (Linux - Jan  8 2025, 13:50:18)
Copyright 2009 - 2025 Dell Inc. All rights reserved.

Usage: cctk --option[=argument]

For more information about a particular command, 
use the option '-H' followed by the command name.
Example: cctk -H --Asset

Note: Commands with * Symbol are Read-Only BIOS Features.

    --ActiveECoresNumber            --ExtIPv4PXEBootTimeout         --ProgramBtn1
    --AdvBatteryChargeCfg           --FanSpdCpuMemZone              --ProgramBtn2
    Advsm                           --FanSpdLowerPcieZone           --ProgramBtn3
    --AllowBiosDowngrade            --FanSpdStorageZone             --PropOwnTag
    --Asset                         --FanSpdUpperPcieZone           --RestoreBIOSSettings
    --AutoOSRecoveryThreshold       --FanSpeedLvl                   -S or search
    --AutoOnPeriod                  --Fastboot                      --SaveConfig
    --BiosAutoRcvr                  --FirstPowerOnDate*             --SecureBoot
    --BiosCharacteristics*          FotaProfile                     --SedBlockSidAuthentication
    --BiosConnectActivation         -H or --help                    --SetupPwd
    --BiosCurLang*                  --HdFreeFallProtect             --ShutdownFailsafeTimer
    --BiosListInstallLang*          --HddInfo*                      --SleepMode
    --BiosRomSize*                  --HddPwd                        --SmmSecurityMitigation
    --BiosVer*                      HttpBootProfile                 --SupportAssistOSRecovery
    --BlinkPsu1Led                  -I or --infile                  --SvcTag*
    --BlinkPsu2Led                  --IrstTimer                     --SysId*
    --BluetoothDevice               --KbdBacklightTimeoutAc         --SysName*
    BootOrder                       --KbdBacklightTimeoutBatt       --SysPwd
    --BrightnessAc                  KeyboardBackLightColor          --SysRev*
    --BrightnessBattery             -L or --logfile                 --TCCActOffset
    --CapsuleFirmwareUpdate         --LastBiosUpdate*               --TypeCPower
    --ChassisIntrusionCounter       --LegacyOrom                    --UefiBootPathSecurity
    CloudAppProfile                 --MasterPasswordLockout         --UsbPortsExternal
    --CompletionCode*               --Mem*                          --Uuid*
    --Computrace*                   --MfgDate*                      --ValSetupPwd
    ConnectionProfile               --MinSizeOfContigMem*           --ValSysPwd
    --CpuCoreExt                    --MpmCfg                        --Version*
    --CpuCount*                     -O or --outfile                 --Virtualization
    --CpuSpeed*                     --OnBoard1394                   --WirelessDevice
    CsosProfile                     --OvrWrt*                       --WirelessLan
    --EmbNic1                       --PeakShiftBatteryThreshold     --WirelessSwitchChanges
    --EmbSataRaid                   --PeakShiftCfg                  --WirelessWwan
    --EnclaveSize                   --PowerWarn                     --WlanRegionCode
    --EnergyStarLogo                --PpiBypassSedBlockSidCommand   
12
```
```
sudo  pkexec /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Adaptive

There was an error setting the option 'PrimaryBattChargeCfg'
echo $?                                                        
43

```

maniacx, 2025-04-06:
Seems like you device does not support  --PrimaryBattChargeCfg
Not sure if it is a dcc issue or your device doesnt support charging threshold.

**Does you laptop have the option to set Charging threshold while running Windows or in Bios?**

Also since you are running the latest kernel, (and if your laptop does support charging threshold), you might  be able to use sysfs although this also is a hit or miss.

```
echo 'Custom' | sudo tee /sys/class/power_supply/BAT0/charge_types
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

OR you might have to try  the older version of dcc
https://www.dell.com/support/home/en-us/drivers/DriversDetails?driverId=MK0CK

luyatshimbalanga, 2025-04-06:
Yes it has Charging threshold currently set to "Adaptive" by default. I haven't tested other mode yet. The device no longer has Windows which has such features.

The above command led to
```
echo 'Custom' | sudo tee /sys/class/power_supply/BAT0/charge_types3 
Custom
tee: /sys/class/power_supply/BAT0/charge_types: Input/output error
```
Downgrading command-configure hardly makes difference

maniacx, 2025-04-06:
> Yes it has Charging threshold currently set to "Adaptive" by default. I haven't tested other mode yet. The device no longer has Windows which has such features.
>
> The above command led to
>
> ```
> echo [...]

Sadly I think your device doesnt support charging threshold.

https://www.dell.com/community/en/conversations/inspiron/limit-battery-charge-level-on-inspiron-14-7445-2-in-1/67e60d001f413b3c1a016755

Ability to set battery charging threshold is a feature provided by manufacturer.
If your device supports it , you should see an option to set charging threshold in your BIOS.

And kernel, libsmbios, dcc will communication with the embedded controller to set the threshold, so that the threshold is applied even when the laptop is powered off.

 BIOS it looks  something like this.
[image]

luyatshimbalanga, 2025-04-06:
Actually the device supports charging threshold although the BIOS user interface differs:

[image]
[image]

Another possible issue may be related to the BIOS but I am unsure

maniacx, 2025-04-07:
> Actually the device supports charging threshold although the BIOS user interface differs:
> 
> Another possible issue may be related to the BIOS but I am unsure

May be some sort of compatibily issue. At this point I am not sure how to help you.

Here are some pointers, that help works on some Thinkpad, Asus laptop that were having issues with charging threshold.
* Resetting BIOS to factory default.
* Hard reset of Embedded controller. This is usually done using by holding power button for 30 seconds, or some older models have a small pinhole with a switch that can be used to reset. Every model have different ways to do this. Search for CMOS / BIOS /EC reset
* Seems like your BIOS is pretty old. You might try to use an even older version of dcc if exist.
* Upgrading BIOS firmware: while some upgrading firmware fixed the issue, but it can go in any direction.

As for the extension, it wont work until you are succesfull in applying charging threshold using command line.
If applying threshold with command line using smbios / dcc / sysfs is successful, then the extension will work.

niluzz, 2025-10-16:
> [@niluzz](https://github.com/niluzz)
> 
> Você pode tentar esses 2 comandos novamente?
> 
> ```
> sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg
> echo $?
> ```
> 
> ```
> sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=Standard
> echo $?
> ```
> 
> ```
> sudo /opt/dell/dcc/cctk --PrimaryBattChargeCfg=PrimAcUse
> echo $?
> ```
> 
> Além disso, se você quiser consultar o manual da linha de comando, pode conferir aqui
> 
> [https://www.dell.com/support/manuals/en-us/command-configure/dcc_5.x_ref_guide/introduction-to-dell-command-%7C-configure?guid=guid-8ed40681-7cf4-4f1e-95bf-d086f3cfadaf&lang=en-us ](https://www.dell.com/support/manuals/en-us/command-configure/dcc_5.x_ref_guide/introduction-to-dell-command-%7C-configure?guid=guid-8ed40681-7cf4-4f1e-95bf-d086f3cfadaf&lang=en-us)
> 
> [dcc-5-x-ref-guide-en-us.pdf ](https://github.com/user-attachments/files/22806153/dcc-5-x-ref-guide-en-us.pdf)

[image]

Aera23, 2026-10-25:
on my Dell Vostro 3450, the same error:
`➜ sudo smbios-battery-ctl -v -c
Libsmbios version : 2.4.3
smbios-battery-ctl version : 2.4.3
 You can use smbios-battery-ctl utility to view/modify battery settings
 Get Charging State : res[smi.cbRES1]=0x0,res[smi.cbRES2]=0x2, res[smi.cbRES3]=0xFF
 
 Supported battery charging features: 
         Charge Disable
 
 Battery charging Status: 
         NIL

 Battery 0 Express Charge State:
         Battery is present, and it does not support express charge`

For reference, here's the basic output: 

`➜ sudo smbios-battery-ctl -v -c
Libsmbios version : 2.4.3
smbios-battery-ctl version : 2.4.3
 You can use smbios-battery-ctl utility to view/modify battery settings
 Get Charging State : res[smi.cbRES1]=0x0,res[smi.cbRES2]=0x2, res[smi.cbRES3]=0xFF
 
 Supported battery charging features: 
         Charge Disable
 
 Battery charging Status: 
         NIL

 Battery 0 Express Charge State:
         Battery is present, and it does not support express charge`

maniacx, 2026-11-09:
@Aera23 @niluzz @luyatshimbalanga @McThump @s4vi0rodrigues @zz0rk 

I just some new information for Dell laptop regarding charging threshold.
https://github.com/maniacx/Battery-Health-Charging/issues/217

I have updated  Github repo with this patch
https://github.com/maniacx/Battery-Health-Charging/commit/a9eb49c00cd8d1b4bb26d063331db99a43f3df85

If you have this sysfs node
```
ls /sys/class/firmware-attributes/dell-wmi-sysman/attributes/PrimaryBattChargeCfg
```

This should work in theory. I do not have dell laptop so I cannot test these patches.
Would you be interested in testing and providing feedback?

niluzz, 2026-11-10:
> [@Aera23](https://github.com/Aera23) [@niluzz](https://github.com/niluzz) [@luyatshimbalanga](https://github.com/luyatshimbalanga) [@McThump](https://github.com/McThump) [@s4vi0rodrigues](https://github.com/s4vi0rodrigues) [@zz0rk](https://github.com/zz0rk)
> 
> I just some new information for Dell laptop regarding charging threshold. [#217](https://github.com/maniacx/Battery-Health-Charging/issues/217)
> 
> I have updated Github repo with this patch [a9eb49c](https://github.com/maniacx/Battery-Health-Charging/commit/a9eb49c00cd8d1b4bb26d063331db99a43f3df85)
> 
> If you have this sysfs node
> 
> ```
> ls /sys/class/firmware-attributes/dell-wmi-sysman/attributes/PrimaryBattChargeCfg
> ```
> 
> This should work in theory. I do not have dell laptop so I cannot test these patches. Would you be interested in testing and providing feedback?

**I tried it but without success, I set a password in the BIOS, it did all the possible tests in different environments but it didn't work, unfortunately. Dell Inspiron 15 3535 fullAMD**

maniacx, 2026-11-11:
> 
> **I tried it but without success, I set a password in the BIOS, it did all the possible tests in different environments but it didn't work, unfortunately. Dell Inspiron 15 3535 fullAMD**

@niluzz 
I forgot to mention. In device settings, for
`Select configuration for charging mode` , select `Sysfs node`

Disable / Renable extension.

niluzz, 2026-11-14:
> > **Tentei, mas sem sucesso. Defini uma senha na BIOS, realizei todos os testes possíveis em diferentes ambientes, mas infelizmente não funcionou. Dell Inspiron 15 3535 fullAMD**
> 
> [@niluzz](https://github.com/niluzz) Eu esqueci de mencionar. Nas configurações do dispositivo, para `Select configuration for charging mode`, selecione`Sysfs node`
> 
> Desativar/Reativar extensão.
I believe it's my notebook model and its BIOS that's different from Intel's BIOS; the extension doesn't have that option. So I think it's something specific to my version. I even used Claude to try and solve it in the best way, but my model, using Linux, only allows adjustments directly in the BIOS and not through the extension.

[image]

Aera23, 2026-11-14:
sadly, I did "tree /sys | grep PrimaryBatt" and couldn't find it... also my BIOS doesn't show battery charge thresholds (I explored all the available menus)

maniacx, 2026-11-16:
@Aera23 
Yes if your Bios does not show any options, it means that laptop does not support this feature.

@niluzz 
I also do think this is a model specific compatibility issue.