maniacx/Battery-Health-Charging #143: Feature request: global settings (sync mode across user accounts)

jojoob, 2025-04-16:
I use different user accounts on the same device. At the moment the charging mode is stored per user and when switching users the charging mode changes accordingly.
I would prefer setting (custom) thresholds as an administrator and store the charging mode globally in a way that it is synced across all users.

What do you think about this change in behavior?

maniacx, 2025-04-17:
I don't think this is possible, but i will find out if there is any way to do this.

If you have some ideas than let me know

jojoob, 2025-04-18:
I have no experience with gnome extension development, therefore I don't know if there is a "gnome extension way" for global settings. The only thing I can think of is to write those settings into a group writable file and read the file on user login.