maniacx/Battery-Health-Charging #144: "Missing dependencies" despite dependency present

axelsimon, 2025-04-17:
I've installed Battery-Health-Charging on a Framework laptop using nix (home-manager) on Fedora Silverblue 41.

Both `gnomeExtensions.battery-health-charging` and `framework-tool` are properly installed.

When trying to enable the extension in Gnome Extensions, i get the Missing dependencies error pop up, but looking at Gnome Shell logs, we see:
```
$ journalctl /usr/bin/gnome-shell
…
framework13 gnome-shell[3042]: Battery Health Charging: path = /var/home/user/.nix-profile/bin/framework_tool
…
```
Any ideas?

maniacx, 2025-04-17:
Unlike `sudo`, `pkexec` invokes commands with a bare minimal `$PATH` environment.

To ensure proper execution across different distributions, I explicitly defined additional system paths where common executables reside. 
For security purposes, I excluded any user-writable directories to avoid security issues.

I wasn't aware about  `/var/home/user/.nix-profile/bin` is this customized by you, or a default added by you.

To get it working, I will have to add the path here.

https://github.com/maniacx/Battery-Health-Charging/blob/fb10df61b48a8f44425230925a1418b0f933f227/lib/helper.js#L17

And 

https://github.com/maniacx/Battery-Health-Charging/blob/fb10df61b48a8f44425230925a1418b0f933f227/resources/batteryhealthchargingctl#L115

Let me know, you can manually edit those files and test it.

axelsimon, 2025-04-17:
Hi @maniacx, thanks for the very quick reply.

`/var/home/user/.nix-profile/bin` is the default path added by the most common Nix installer, [DeterminateSystems/nix-installer](https://github.com/DeterminateSystems/nix-installer/) (or in my case the Lix installer, but same thing as it's a fork of the DetSys one)

It's not Fedora Silverblue specific, as in Fedora (Silverblue or otherwise) does not use Nix by default.
What is Silverblue specific is that `$HOME` is `/var/home/user` as that's one of the few mutable dirs in Silverblue. So to your point, that is a user-writable directory.

I've just looked at editing the files, and while it's obvious to me what to add in [Battery-Health-Charging/resources/batteryhealthchargingctl](https://github.com/maniacx/Battery-Health-Charging/blob/fb10df61b48a8f44425230925a1418b0f933f227/resources/batteryhealthchargingctl#L115) i'm not sure about [Battery-Health-Charging/lib/helper.js](https://github.com/maniacx/Battery-Health-Charging/blob/fb10df61b48a8f44425230925a1418b0f933f227/lib/helper.js#L17).
Could it be in [Battery-Health-Charging/lib/helper.js](https://github.com/maniacx/Battery-Health-Charging/blob/fb10df61b48a8f44425230925a1418b0f933f227/lib/helper.js#L69)?