maniacx/Battery-Health-Charging #149: Tuxedo change the the pathes with version 4.14.0

mokilcde, 2025-06-20:
Hi!

tuxedo_keyboard changes the paths from 
`/sys/devices/platform/tuxedo_keyboard/charging_profile/charging_profile`
and
`/sys/devices/platform/tuxedo_keyboard/charging_profile/charging_profiles_available`
to
`/sys/devices/platform/tuxedo_keyboard/charging_priority/charging_prio`
and
`/sys/devices/platform/tuxedo_keyboard/charging_profile/charging_profiles_available`

maniacx, 2025-06-20:
Also what is you make an model of you laptop kernel version numbe

```
cat /sys/devices/virtual/dmi/id/board_name
cat /sys/devices/virtual/dmi/id/product_family
cat /sys/devices/virtual/dmi/id/product_name
cat /sys/devices/virtual/dmi/id/sys_vendor
uname -r
```

maniacx, 2025-07-20:
Hello @mokilcde 

Any updates?

closed 2025-09-05 by maniacx