maniacx/Battery-Health-Charging #152: Add support for other Acer dependencies, or is there a way to change it?

dicktivity, 2025-07-25:
Hello, my device is Acer Nitro AN515-45 using Zorin OS 17.3 Core and for some reason i can't get [https://github.com/frederik-h/acer-wmi-battery/](url) working. But fortunately [https://github.com/maxco2/acer-battery-wmi](url) is working perfectly, i believe the GUID are identical (cmiiw). Now it would be nice if i can have this extension working with mxco2's module instead of using terminal.

dicktivity, 2025-07-26:
Got an error
[image]
```
The settings of extension Battery-Health-Charging@maniacx.github.com had an error:
SyntaxError: import declarations may only appear at top level of a module

Stack trace:

_init@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:23:33
ExtensionPrefsDialog@resource:///org/gnome/Shell/Extensions/js/extensionPrefsDialog.js:10:4
OpenExtensionPrefsAsync@resource:///org/gnome/Shell/Extensions/js/extensionsService.js:124:33
async*_handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:368:35
_wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:401:34
run@resource:///org/gnome/Shell/Extensions/js/dbusService.js:186:20
main@resource:///org/gnome/Shell/Extensions/js/main.js:22:13
run@resource:///org/gnome/gjs/modules/script/package.js:206:19
start@resource:///org/gnome/gjs/modules/script/package.js:190:8
@/usr/share/gnome-shell/org.gnome.Shell.Extensions:1:17
``` 

And when i log out it got replaced with [gnome extension version](https://extensions.gnome.org/extension/5724/battery-health-charging/)
I don't get that error when installing from [gnome extension](https://extensions.gnome.org/extension/5724/battery-health-charging/), instead i got dependencies missing error

maniacx, 2025-07-26:
Which version of Gnome?
```
gnome-shell --version
```

dicktivity, 2025-07-26:
GNOME Shell 43.9

maniacx, 2025-07-26:
Try this branch

https://github.com/maniacx/Battery-Health-Charging/tree/GNOME42-44

dicktivity, 2025-07-26:
IT WORKED!
[image]

at first it give a polkit error, turns out i just have to click install privileges for this user and input my password

[image]

Thank you so much, appreciate your assistance.