maniacx/Battery-Health-Charging #155: GalaxyBook support

maniacx, 2025-07-28:
@Alzemand 

If you have replaced it with correct path, it should work.
Do note that you need to logout and login to apply the updated code.

Also can you give me the output of the command
```
ls /sys/module | grep -i galaxy

ls /sys/class/firmware-attributes
```

Alzemand, 2025-07-28:
Yes, I install and logout e login. And get:

Charging threshold not updated.

maniacx, 2025-07-28:
May be it accepts only 85?

Disable extension.
And try different values using commandline

```
echo '100'  | sudo tee '/sys/class/power_supply/BAT1/charge_control_end_threshold'

cat /sys/class/power_supply/BAT1/charge_control_end_threshold

echo '80'  | sudo tee '/sys/class/power_supply/BAT1/charge_control_end_threshold'

cat /sys/class/power_supply/BAT1/charge_control_end_threshold

echo '85'  | sudo tee '/sys/class/power_supply/BAT1/charge_control_end_threshold'

cat /sys/class/power_supply/BAT1/charge_control_end_threshold

echo '60'  | sudo tee '/sys/class/power_supply/BAT1/charge_control_end_threshold'

cat /sys/class/power_supply/BAT1/charge_control_end_threshold

echo '100'  | sudo tee '/sys/class/power_supply/BAT1/charge_control_end_threshold'

cat /sys/class/power_supply/BAT1/charge_control_end_threshold

```

Try different values.

Also check logs

Open console and use command

```
journalctl -f -o cat /usr/bin/gnome-shell
```
Do not close console
Enable extension. See if there are any errors.

maniacx, 2025-07-29:
Great. I will release an update this week. Thank you.

maniacx, 2025-08-04:
@Alzemand 

The developer of Galaxy Book shared some information with me.
It appears you are using an older, standalone driver, probably compiled and installed from GitHub.
This driver is already included in recent Linux kernel versions, so if you are running a newer kernel, you do not need to install a separate kernel module from GitHub.

The kernel’s built-in module is slightly different from the one on GitHub. That’s why my initial extension code did not work correctly.

I have now made some changes so that the extension supports both the older driver and the newer built-in driver.
Could you please test the extension again before I release it?

Thank you.

Alzemand, 2025-08-10:
Yes, I use a compiled driver in my kernel, but I will test it on the new kernel. I read that this driver has been included in newer kernels, and I will test it for us.