maniacx/Battery-Health-Charging #156: Asus Vivobook 14 notebook does not work

mkevison, 2025-08-01:
Unable to limit charging on an Asus notebook with an Intel Core Ultra 7 258V processor

Always charging above the configured limit

maniacx, 2025-08-01:
Do some test.
Disable the extension and change threshold via sysfs, using command line

Check your power supply name for battery. Could be BAT0, BAT1, BATC or BATT
```
ls /sys/class/power_supply/
```

Example if BAT0 is found under power supply names use it and  a value 60
```
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```
Verify the threshold is applied by reading. This should display current values.
```
cat /sys/class/power_supply/BAT0/charge_control_end_threshold
```
Now verify you charging status looking at Battery Indicator or command
```
cat /sys/class/power_supply/BAT0/status
```

Try different values and check if it responds.

Let me know what you find.

---
If the issue happens only when in sleep mode or hibernate enable this settings below

[image]

Test using this Enabled.

nadkaro, 2025-08-22:
ASUSTeK COMPUTER INC. P2540UV
---
It worked for me once and then it stopped working

[image]

maniacx, 2025-08-22:
Seems like charging_control_end_threshold is not exposed anymore.

This is an issue with kernel or kernel configuration.

For some features to work such as charging_threshold, keyboard backlight etc. `asus_wmi ` kernel module needs to be loaded.

Check if it is loaded

```
lsmod | grep "asus"
```

You should see asus_wmi in the list

[image]

nadkaro, 2025-08-22:
Yes

[image]

maniacx, 2025-08-24:
@nadkaro 

Kernel module is loaded.

Does you laptop support charging threshold in windows? You did mention, the extension was working earlier.

1. May be a firmware update, made it incompatible, with kernel or corrupted some embedded controller (EC) values during firmware upgrade.
2. Or kernel recently updated with a patch that is causing issue with your laptop.

Soft reset of Bios. 
Sometime resetting Bios  to `Restore default`  / `Load optimize settings` helps. Not

Hard reset of bios
Power off laptop
Connect the power adapter and plug it into a nearby wall outlet.
Press and hold the Power button for 40 seconds. 

Note: Doing the procedure above, you might lose all settings save in Bios, such as Boot priority.

Anyways this is not a extension issue. But hardware / kernel issue