maniacx/Battery-Health-Charging #158: Giving out "Missing dependencies error"

KayWhyDie, 2025-08-21:
Can use the kernel module on its own but the extension doesn't work.

KayWhyDie, 2025-08-25:
Even though I managed to fix the dependency error, trying to make it work with polkit ended up breaking things. I tried fixing it myself but couldn’t find a proper workaround. Later, I asked GPT-5 to help me merge everything I had done into a clearer and more reliable step-by-step process. After many attempts, the only workaround that worked on my machine was the following:

# ================================================
# Battery Health Charging fix for AORUS laptops
# (Debian/Ubuntu Wayland, polkit >=124 bug workaround)
# ================================================

# -----------------------------
# 1. BACKUP EXTENSION
# -----------------------------
cp -a ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com \
      ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com.bak.$(date +%F-%H%M)

# -----------------------------
# 2. DISABLE EXTENSION WHILE PATCHING
# -----------------------------
gnome-extensions disable Battery-Health-Charging@maniacx.github.com

# -----------------------------
# 3. PATCH GIGABYTE → AORUS PATHS
# -----------------------------
# Fix devices/Gigabyte.js
sed -i 's#/sys/devices/platform/gigabyte_laptop#/sys/devices/platform/aorus_laptop#g' \
  ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/devices/Gigabyte.js

# Fix resources/batteryhealthchargingctl
sed -i 's#/sys/devices/platform/gigabyte_laptop#/sys/devices/platform/aorus_laptop#g' \
  ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources/batteryhealthchargingctl

# Ensure AORUS kernel driver loads at boot
echo "aorus_wmi" | sudo tee /etc/modules-load.d/aorus_wmi.conf

# -----------------------------
# 4. INSTALL HELPER FOR SUDO (WORKAROUND FOR POLKIT BUG)
# -----------------------------
# Copy helper script to /usr/local/bin
sudo cp ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/resources/batteryhealthchargingctl \
        /usr/local/bin/batteryhealthchargingctl-custom
sudo chmod 755 /usr/local/bin/batteryhealthchargingctl-custom

# Allow running helper via sudo without password (replace USERNAME)
echo "USERNAME ALL=(ALL) NOPASSWD:/usr/local/bin/batteryhealthchargingctl-custom" | \
  sudo tee /etc/sudoers.d/batteryhealthcharging-custom
sudo chmod 440 /etc/sudoers.d/batteryhealthcharging-custom

# -----------------------------
# 5. PATCH EXTENSION TO USE SUDO INSTEAD OF PKEXEC
# -----------------------------
# driver.js
sed -i "s/\['pkexec',/['sudo',/" \
  ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/driver.js

# helper.js
sed -i "s/\['pkexec',/['sudo',/" \
  ~/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/lib/helper.js

# -----------------------------
# 6. RELOAD GNOME SHELL
# -----------------------------
# On Xorg:   Press Alt+F2, type r, hit Enter
# On Wayland: log out and back in

# -----------------------------
# 7. RE-ENABLE EXTENSION
# -----------------------------
gnome-extensions enable Battery-Health-Charging@maniacx.github.com

# -----------------------------
# 8. TEST
# -----------------------------
# Change charge limit in GNOME extension UI, then check:
cat /sys/devices/platform/aorus_laptop/charge_mode
cat /sys/devices/platform/aorus_laptop/charge_limit

closed 2025-08-25 by KayWhyDie

maniacx, 2025-08-25:
I apologize.  I didn't pay attention to your first post. I completely missed the path was not correct.

aorus_laptop instead of gigabyte_laptop.

Thanks. I will add the path to my extension.