maniacx/Battery-Health-Charging #162: After an update on openSUSE Tumbleweed (polkit 126), the extension keeps prompting for a password every time

kbdragon, 2025-09-13:
After updating openSUSE Tumbleweed (polkit updated to polkit-126-2.1.x86_64), the Battery Health Charging extension started misbehaving.

On login, the extension doesn't work and I'm prompted for a password:

[image]

I managed to get the extension working by manually disabling and re-enabling it, but I get prompted for a password on every state change.

kbdragon, 2025-09-13:
@maniacx 
It's a Lenovo IdeaPad 3 (15ITL6 Type 82H8)
Kernel 6.16.6-1-default
GNOME Shell 48.4
I tried poking around to see what went wrong, but I'm not knowledgable enough to understand it. Happy to follow instructions, though, if you'd like me to check anything.

maniacx, 2025-09-13:
I didnt notice the check time out error.

This is related polkit bug for ubuntu.
The workaround is to use sudo. 

But there was a conversation about this bug being solved in polkit 126. However I didn't test it.

I will install opensuse tumbleweed today to check.

Workaround to use sudo

https://macs18max.github.io//Battery-Health-Charging/polkitbug

Discussion about bug

https://github.com/polkit-org/polkit/issues/572

https://github.com/polkit-org/polkit/issues/577

I think it is still broken in 126 as per issue 577

maniacx, 2025-09-14:
## **Bug while changing threshold using pkexec**
The extension installs polkit rules under /etc/polkit-1/rules.d for allowing pkexec to run without password prompt
After adding some debugging flags to polkit and restarting polkit.service, I found on opensuse, I found that rules from  /etc/polkit-1/rules.d are not applied.

```
Stopping Authorization Manager...
Sep 13 21:00:38 localhost.localdomain systemd[1]: polkit.service: Deactivated successfully.
Sep 13 21:00:38 localhost.localdomain systemd[1]: Stopped Authorization Manager.
Sep 13 21:00:38 localhost.localdomain systemd[1]: polkit.service: Consumed 2.019s CPU time.
Sep 13 21:00:38 localhost.localdomain systemd[1]: Starting Authorization Manager...
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Started polkitd version 126
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Loading rules from directory /etc/polkit-1/rules.d
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Error opening rules directory: Error opening directory \E2\80\9C/etc/polkit-1/rules.d\E2\80\9D: Permission denied (g-file-error-quark, 2)
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Loading rules from directory /run/polkit-1/rules.d
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Error opening rules directory: Error opening directory \E2\80\9C/run/polkit-1/rules.d\E2\80\9D: No such file or directory (g-file-error-quark, 4)
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Loading rules from directory /usr/local/share/polkit-1/rules.d
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Error opening rules directory: Error opening directory \E2\80\9C/usr/local/share/polkit-1/rules.d\E2\80\9D: No such file or directory (g-file-error-quark, 4)
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Loading rules from directory /usr/share/polkit-1/rules.d
Sep 13 21:00:38 localhost.localdomain polkitd[14527]: Finished loading, compiling and executing 6 rules
```

This is due to directory permission set by opensuse for /etc/polkit-1/rules.d

Comparing both Opensuse and my Fedora permissions

Opensuse
```
ls -ld /etc/polkit-1/rules.d
drwxr-x---. 1 root root 154 Sep 13 20:35 /etc/polkit-1/rules.d
```

Fedora
```
drwxr-x---. 2 root polkitd 4096 Sep 13 20:05 /etc/polkit-1/rules.d
```

You could ask opensuse if this directory ownership as root and not polkitd is intentional or a bug.

## **Bug during installation of polkit rules**
Also, for some reason I get permission denied for installer.sh under, /home/$USER/.local/share/gnome-shell/extensions/Battery-Health-Charging@maniacx.github.com/tool/installer.sh
If I place the same file in example /home/$USER/Downloads/installer.sh, it executes. I could not find the reason for permission denied